simplefoc/Arduino-SimpleFOCShield #2: Could you please provide the altium project of v1.3.3？

candyriver, 2020-11-30:
the altium project in the release zip is still v1.3.2.

askuric, 2020-12-07:
Hey @candyriver 
You are actually right, sorry about that. I have actually forgotten to save this version and have continued developing the version 2.0 which is in the dev branch now. It will come out very soon.

Version v1.3.3 is very similar to the 1.3.2 and there are just two or three small changes that can easily be integrated to the altium version v1.3.2. 

I am sorry about this, but I will not be able to update the v1.3.3 altium project. :D

askuric, 2020-12-09:
Hey @candyriver , 
Due to the recent error that has been revealed to me I had to recreate the v1.3.3 Altium model and the new v1.3.3 release is fixed. You can find the altium model for v1.3.3 in the repository. 
Cheers,
Antun

closed 2020-12-14 by askuric