simplefoc/Arduino-SimpleFOCShield #4: INA240: which gain should be selected?

gowitty, 2021-03-07:
there 4 different gains.  which gain should be selected?
INA240A1 GAIN(V/V) 20
INA240A2 GAIN(V/V) 50
INA240A3 GAIN(V/V) 100
INA240A4 GAIN(V/V) 200

askuric, 2021-03-07:
Hey @gowitty,

So in short, I've taken a choice to use:
- Shunt resistor value: **0.01Ohm**
- Current sense: **INA240A2 (gain 50)**

#### Here is why:

So in general there are two parameters when it comes to current sensing hardware:
 - shunt resistor values
 - amplifier gain

In the case of the *Simple**FOC**Shield*, the BLDC driver chip L6234 has the current limit of 5 Amps. So we need to be able to measure +-5amp current. But in most cases your application with the *Simple**FOC**Shield* will remain in the 3Amp range. Higher currents require appropriate heat evacuation: heat-sinks, fans, etc. 

So you need to be able to measure +-3amps for example, which is a 6Amp range.  (or +- 5amps -> 10amp range).
The mcus usually have the 3.3V or 5V logic which tells you that you need the transformation ratiofor example:
```
trans_ratio = 5V / 10amps = 0.5 V/Amp    or    trans_ratio = 3.3V / 10amps = 0.33 V/Amp 
```

So now that we know what is the transformation ratio that we need we can proceed to finding the shunt resistor value and the amplification gain. We know that the `trans_ratio = amp_gain*shunt_resistance`. Now there is a trade-off here in between amplification gain and shunt value:
Shunt resistance will be placed in-line with the motor and all the phase current of the motor will pass through the shunt resistor so it is important to choose the value that is the smallest possible value to reduce the heat loss. Additionally by adding the shunt resistors in-line with motors we are actually increasing the phase resistance of the motor, and since we are placing shunt resistors only on two phases this can make the motor unbalanced and this is another reason that makes it really important to keep this value small relatively to the phase resistance of the motor. On the other hand, the smaller the shunt resistor value the smaller the voltage across it that we want to measure. Since the voltage across it is smaller it which makes it more difficult to measure the phase current, because the smaller it is the higher the amplification noise.  So we would prefer to have the shunt resistor value high and amplification gain low to reduce the noise. 

Since we know that the phase resistances of the BLDC motors to be used with this board will be in the range of >10Ohm we know that we can take the shunt resistor of >0.01Ohm. 
Now we can calculate that for shunt resistor of 0.01 Ohm we would need the amplificaiton gain:
```
amp_gain= 0.5/0.01 = 50     or     amp_gain =  3.3/0.01 = 33
```
So in order to measure the +-5Amps on 5V logic we can use INA240A2 (gain 50)  but on the 3.3V logic that will give us +-3.3Amp range. 
If we want to have higer range on 3.3V logic we need to either:
- use INA240A1 (gain 20) which will give us +-8.25Amp range on 3.3V and +-12.5Amp range on 5V.
- use INA250A2 (gain 50) and change the shunt value (ex. 6 miliOhm) which would give use +-5.5Amps range on 3.3V and +-8.5Amps range on 5V

So I've decided to stick with the **0.01Ohm** shunt and **INA240A2** because even +-3.3Amp range is perfectly enough for gimbal motors.

gowitty, 2021-03-13:
Hey @askuric 
appreciate sharing so many details!

closed 2021-03-13 by gowitty