simplefoc/Arduino-SimpleFOCShield #5: Header pins too tall?

steph643, 2021-06-08:
On the left, a nucleao-64 with its 8mm pins. On the right, my brand new SimpleFOCShield with its 18.5mm pins, which cannot penetrate fully into the 9mm Arduino header.
[image]
[image]
Is it unintended or am I missing something?

closed 2021-06-08 by steph643

steph643, 2021-08-08:
> I've pivoted from them, for the newest batch though. Even they were out of stock

I'm fighting to find the same kind of connector (in double-row). Would you disclose who your supplier is?

askuric, 2021-08-09:
Hey @steph643,
I've tried many of them and I am buying them online from the webshops as well. Because my quantities are small.

So I've started with the mouser, where I've bought every component separately. But they soon rum out of components. 
Then I've bought them from spark fun. But they to run out.
Then Arduino webshop.
The last one I've bought was from afafruit. 

And then even the other components disappeared from the market 😃