simplefoc/Arduino-SimpleFOCShield #7: confuse about pin mux

standback, 2021-12-20:
hi guys.
i have question about  "Customizing pinout",
In the docs. 
[image]
pwmA mapping to pin 9/10
pwmB mapping to pin 3/5

But , according the  note below
`
Now, there is a lot of possible pin configurations and not all of them are possible depending on the microcontroller and sensor you are using. For example, Arduino UNO only has 2 external interrupt pins, and they are pin 2 and 3. Therefore when using the board with Arduino UNO and Encoder we will try to use pin 3 for encoder channel A and not for pwm A.`
it seems mean that  PwmA should  mapping to pin3  instand 9/10.
version2.03 board ,also design that pinA mapping to pin 3/5.
[image]

is that means  this pinmux is  out of date?

I am confused. 
[image]
,
[image]
which one is right?

askuric, 2021-12-20:
Yeah, it's the docs. It was an example pin out for the earlier version of the board.
Whatever is indicated on the board itself is the right configuration.

We try to keep the docs as up-to-date as possible, but these kinds of stuff are sometimes difficult to detect :)

We'll update it as soon as possible.

standback, 2021-12-20:
Got it,  thanks for your reply.

closed 2021-12-20 by standback