simplefoc/Arduino-SimpleFOCShield #9: I2C pullups using ESP32-S2 Metro

wasistdalos, 2023-10-14:
Hi, thank you a lot for the project, the documentation and effort! using the ESP32-S2 Metro there are already I2C pullups included so i didn't connect the jumper on the SimpleFOC V2.0.4. Using that setup with an AS5048b i did run into issues getting I2C communication to work.  Whitout the shield it did work as expected. I solved the issue by desoldering one of the 4,7k I2C pullups of the V2.0.4 shield.  Please check for next shield version, thank you! 
[image]

askuric, 2023-10-14:
Hey @hammerflo,

This is very strange! :thinking: 

Did you try to desolder the connection (solder pad) on the bottom side of the board?
There is one:
[image]

hammerflo, 2023-10-14:
yes i did. looking at the schematic, sda and scl are connected with 2x 4,7k on the shield. maybe this leads to this behaviour?

askuric, 2023-10-14:
Nice catch!
This could be an issue, I did not think about that. 
And in that case it make sense that desoldering one of the resistors solved the problem.

I've connected their 5V channel to reduce the number of soldering pads in the v2.0.4. Earlier versions did not have this issue. I'll make sure to separate them in the next release.

askuric, 2024-06-26:
The issue is resolved in the new version of the shield :D